tum-ei-eda/etiss #11: RISCV bootup

uzleosharif, 2020-04-21:
Currently in both the 32/64 bit versions of RISCVArch implementations, we are forced to boot from 0x80 which might not be true for general RISCV SoCs, though compatible with Pulpino. We have to provide flexibility and remove this hardcoded impl.

Moreover it also seems that **RISCArch::resetCPU(... cpu, ... startpointer)** doesn't really respect the custom passed startpointer and this could explain why earlier we had to hack the CPU to start from 0x80. So maybe the starting point to fixing this issue would be get **RISCArch::resetCPU(...)** working for custom start PC

uzleosharif, 2020-04-21:
just checked the reset interface is also not working for other targets as well so more accurately it is an ETISS issue with nothing specific to RISCV

rafzi, 2021-01-20:
the entry point is now chosen based on the loaded ELF file

closed 2021-01-20 by rafzi