tum-ei-eda/etiss #30: config value precision loss

rafzi, 2020-12-01:
We use "GetDoubleValue" from the SimpleINI library to get integer config values that can be larger than 32 bit.

This is an issue because very large values will be skewed to the nearest representable double, changing the intended value silently.

I'd suggest to patch our own "GetLongLongValue" into the library, since we keep a copy of it in this repository anyways.

wysiwyng, 2021-09-16:
Commit 1e6d15c0d7baa527a70cb986a3c2a7b0125f552d changed that to just converting the value string in the .ini file "by hand", without using SimpleIni's `Get*Value` functions. Changing `stod` to `stoll` in line 520 would fix this.

wysiwyng, 2021-12-07:
fixed in #96.

closed 2021-12-07 by wysiwyng