tum-ei-eda/etiss #41: LLVM 11 needs C++14 enabled

wysiwyng, 2021-01-22:
Currently when compiling ETISS with LLVM enabled a rather unwieldy template error related to various smart pointer features is thrown during compilation of `LLVMJitLib.cpp`. Setting the C++ standard to 14 in CMakeLists.txt fixes this.

No pull request to put this up for discussion, as there were some issues with C++14 in the past.

rafzi, 2021-01-26:
I have tried upgrading to C++14 on the "cpp14" branch. Everything looks fine. Was there some other issue or could we just merge that?

wysiwyng, 2021-01-26:
I personally have no points against that move, I just remember there being some talk about compatibility issues with e.g. some MINRES libraries.

JoGei, 2021-01-26:
I tried C++17 sometime last week, too. If we do the upgrade (and possibly fix a few things when encountering them) why do the bigger leap?

rafzi, 2021-01-26:
> I personally have no points against that move, I just remember there being some talk about compatibility issues with e.g. some MINRES libraries.

as far as i understood, that was just an issue for the vp library because it might use minres stuff.

> I tried C++17 sometime last week, too. If we do the upgrade (and possibly fix a few things when encountering them) why not make the bigger leap?

fine with me as well. ubuntu 18.04 has gcc 7.4 which is feature complete. could you put up this upgrade as a PR?

JoGei, 2021-01-26:
> as far as i understood, that was just an issue for the vp library because it might use minres stuff.

Yes this does not affect ETISS

> fine with me as well. ubuntu 18.04 has gcc 7.4 which is feature complete. could you put up this upgrade as a PR

Ok. AFAIK it was just removing the forced c++11 in https://github.com/tum-ei-eda/etiss/blob/6f2750927083d492cf974f14686a062cccb2ab95/CMakeLists.txt#L67

closed 2021-01-29 by rafzi