tum-ei-eda/etiss #48: command line options

uzleosharif, 2021-02-04:
We can currently pass in config options using ini scripts or as command-line arguments. Currently we allow a certain options to be overwritten. For example we can pass 2 ini files:
```
// foo.ini
[Int Configurations]
a=0

// bar.ini
[IntConfigurations]
a=4
b=3

```

when we run etiss passing in these 2 files
```
> <etiss program> -i foo.ini -i bar.ini
```

then in this case `etiss::cfg` would take `a` as 0 and ignore overwrites

How should we deal such overwriting scenarios. I guess the easiest solution would be to disable this feature altogether and the user should make sure that he is not passing in conflicting values for a setting.

uzleosharif, 2021-02-04:
For reference, current ETISS handles such overwriting options in the following manner:
* for `IntConfiguration` **first** seen value is passed to `etiss::cfg`
* for 'BoolConfiguration` **first** seen value is passed to `etiss::cfg`
* for `StringConfiguration` **last** seen value is passed to `etiss::cfg`

within the program you can always override a given setting using `etiss::cfg<...>.set(...)` interface

JoGei, 2021-02-04:
Naturally I'd assume from a command line program that the last argument of the argv[] overwrites any previous ones. So staying with your example I would be surprised that `a!=4` since `bar.ini` was passed "after" `foo.ini`
In addition: `> <etiss program> -i foo.ini -i bar.ini -a 5 -b 4` should mean that the settings for `a` and `b` in `foo.ini`, `bar.ini` are overwritten since the "in-line" command line arguments came after them? Naturally the "in-program" setting would always come last and be dominant.

Another idea could be, to specify the priority of settings by type: For example "in-program">"in-file">"in-line" but I see issues here with then having multiple occasions of the same argument target in the same type, like you gave with your example.

uzleosharif, 2021-02-05:
hi sorry for causing confusion..yes from the command-line all options would be received by `etiss::cfg` e.g. in above example both `a=5` and `a=4` will be seen. Commandline interface via `argc/argv` doesn't filter one. Its within `etiss::cfg` we are deciding which overriding option to take as final value. This issue is this behavior is not consistent as I showed earlier, for some you take the first while for others you take the last. To avoid this behavior, I am arguing for generating an error in case multiple options are encountered at `etiss::cfg` and force the user to pass in the options properly

JoGei, 2021-02-05:
Personally, I would find the option to overwrite settings of a passed input file with "in-line" arguments quite convenient. But yes, just restricting multiple settings for the same variable would at least be a well defined behavior.

closed 2021-03-17 by rafzi