tum-ei-eda/etiss #52: SimpleMemSystem ignores empty ELF segments

wysiwyng, 2021-02-15:
See title. Empty segments might result from minimal programs where the `.data` and `.bss` segments are left empty by the compiler but should still get allocated for heap and stack use.

JoGei, 2021-02-15:
Currently the ELF-Interpreter/loader in SimpleMemSystem only interprets ELF Segements, i.e., program headers (see loop over ELFIO::reader::segments [here](https://github.com/tum-ei-eda/etiss/blob/ea71e6191877d9f162227cfa3191b64dcafb53f7/src/SimpleMemSystem.cpp#L112)). 
If you want to include memory sections defined outside of what the executable "needs", we might need to implement some form of algorithm including the ELF sections, too. However, there is nothing we can generate that does not also specify its size in the ELF.

Do you have a minimal example ready? Then I would update the SimpleMemSystem to support that behavior.

wysiwyng, 2021-02-15:
Sorry about the confusing terminology, `.data` and `.bss` are of course sections. I think what I need is that the elfloader does not skip ELF segments which have a non-zero "memory size" but a zero "file size", i.e. are not filled by anything by the compiler. Currently the reason why "empty" segments are skipped is this if statement:
https://github.com/tum-ei-eda/etiss/blob/ea71e6191877d9f162227cfa3191b64dcafb53f7/src/SimpleMemSystem.cpp#L156 
which checks whether there is any data in a segment. Simply removing the second condition does the trick for me in my case, but might create other issues.

A different mechanism of specifying memory sizes is anyways a good idea to have, especially if you cannot modify the linker script to suit ETISS' needs.

PR #54 (2021-02-18): Allow loading of empty memory segments
Fixes #52.
commit d8031f83 (2021-02-18): Allow loading of empty memory segments

Fixes #52.
closed 2021-02-18 by rafzi