tum-ei-eda/etiss #58: InstructionDefinition with missing callback causes FATALERROR

wysiwyng, 2021-03-26:
Title. The error message which is printed in [Instruction.h](https://github.com/tum-ei-eda/etiss/blob/32ddb52052df07f4241dfc7fda64fa671b72acd1/include/etiss/Instruction.h#L990) suggests that `InstructionDefinition`s with missing callbacks will be ignored, but not the termination of ETISS. What is the desired behavior here?

rafzi, 2021-06-28:
Do they always have a callback in our current definitions?

If there is a use-case where it makes sense that they don't have one, it should be fine to allow this. Of course would have to be tested if everything else can deal with that case.

I don't know the historical reason why this was treated as fatal.

wysiwyng, 2021-06-28:
Our current definitions always have a callback, this question arose from testing things for issue #59. Here I wanted to create minimal examples to keep clutter minimal, but ran into the problem that ETISS does not accept missing callbacks despite the error message stating otherwise.

It probably doesn't make much sense to allow missing callbacks other than for testing, I would just update the error message to reflect that.

commit b826322a (2021-09-15): Update Instruction.h

don't suggest an InstructionDefinition will be ignored when it doesn't have a callback. closes #58.
closed 2021-09-15 by wysiwyng