tum-ei-eda/etiss #69: cmake: copy without depdendency

rafzi, 2021-07-29:
In the main cmake, there is a block of code that copies includes into the build folder for an in-tree build of etiss.

https://github.com/tum-ei-eda/etiss/blob/master/CMakeLists.txt#L296

This is missing some dependency to the original includes. If they are changed, the change will not be reflected in the build folder, unless cmake is run manually.

To fix this we'd probably want to add custom commands: https://stackoverflow.com/questions/8434055/cmake-copy-if-original-file-changed

@uzleosharif

uzleosharif, 2021-07-29:
hi thanks for mentioning the issue. i have a look over the weekend to get to know the issue and possibly try your suggested fix to resolve this issue

uzleosharif, 2021-09-02:
hi @rafzi  i dont remember why we needed to copy include files in the first place..in the latest master branch commit, i am able to use etiss in build-tree even if i dont copy the include files.....so we might as well remove the lines 298-301

however i am not sure  about this issue then if we dont have dependency on <build-dir>/include at all. Could you elaborate with a test example further explaining this issue

rafzi, 2021-09-02:
If the copy is not required, we should remove it and that would resolve this issue.

I'm not sure if there was an actual problem caused by this. I was just noticing that there were these include files that could become out-of-sync with the original ones.

uzleosharif, 2021-09-04:
cool then i remove it and update the master..thanks a lot

uzleosharif, 2021-09-04:
fixed in commit 4c3631391c81ef49d292e030c09ffd085cb3c70c

closed 2021-09-04 by uzleosharif