tum-ei-eda/etiss #101: Better integration of third party libraries

JoGei, 2021-12-07:
Third party integration in ETISS is a bit intransparent. A lot of libraries are copy-pasted into ETISS source code which is mostly fine, but makes is harder to maintain and differ what foreign code is used (e.g., versioning, missing copyrights/licenses). A few examples, I am aware of:
- [ ] [pugixml](https://github.com/tum-ei-eda/etiss/tree/master/src/fault/xml) - used in fault injection
- [ ] [softfloat](https://github.com/tum-ei-eda/etiss/tree/master/src/fpu) - kind of anyway a duplicate for the src/jitlib/softfloat
- [ ] [simpleini](https://github.com/tum-ei-eda/etiss/tree/master/src/IntegratedLibrary/SimpleIni) - extensive use in ETISS runtime configuration

A possible solution to this could be to add a `third_party/` directory to ETISS' root to integrate such things - either through git submodules or CMake's FetchContent. Please feel free to extend the list above, if you are aware of more.

wysiwyng, 2021-12-07:
I would possibly add [tcc](https://github.com/tum-ei-eda/etiss/tree/master/JITImpl/TCC) to this list, currently it is downloaded and built using CMake's `execute_process` interfaces, which is arguably not "the best way to do it"™. I have no idea how this could be made nicer, though.

JoGei, 2022-12-16:
simpleini fixed with f53752f66621b07583ff64efd6d236af550b17d5

JoGei, 2022-12-16:
tcc fixed with #125

wysiwyng, 2022-12-16:
As far as I see it, the mentioned `softfloat` is only used for the `OR1K` core, which probably can be deprecated anyways.

JoGei, 2022-12-16:
pugixml: would be fixed with merge of #100

JoGei, 2023-02-23:
Opened a Draft PR: #132