tum-ei-eda/etiss #136: Dealing with EXIT code of target

PhilippvK, 2023-06-19:
The exit code of the simulated program is either the return value of the `main()` function or the argument to the `exit(-1)` function. For unit testing and benchmarks, we should make sure to not ignore if the program reports a non-zero exit code.

**Previous approach (without Semihosting):**
- Custom `_exit` hook implemented in `syscalls.c`: https://github.com/tum-ei-eda/etiss/blob/40c0b920cf4072adf189211ca44291534355dcfd/examples/SW/riscv/cmake/pulpino_tumeda/syscalls.c#L113-L121
- Based on `printf` -> ROM size overhead -> Only enabled in DEBUG mode 

**Current approach (with Semihosting based on https://github.com/wysiwyng/etiss/tree/coredsl_exceptions)**
- No need to implement custom `syscalls.c` anymore.
- `SYS_EXIT` currently just tells ETISS to exit, ignoring the argument. See: https://github.com/wysiwyng/etiss/blob/a09aea80a433d6b35a2df82176a4f6e39a8fa3e5/src/jitlibs/semihost/semihost.cpp#L405

**Proposed Improvements:**
- Print exit code in handler for `SYS_EXIT` using etiss logger
- Log level Warn/Error for non zero (FAIL) values
- Log level Verbose/Info elsewhere (OK)
- *Optionally* pass through exit code to bare_etiss_simulator to fails as well (if configured by the user)

wysiwyng, 2023-08-22:
`bare_etiss_processor` currently also uses exit codes to signal when ETISS encountered some internal error. Distinguishing these from the application's exit code would not be possible, this would at least have to be noted somewhere. How do other simulators handle this?

PhilippvK, 2023-08-22:
RISC-V OVPSim has a command line flag to choose whether a non-zero `_exit()` call will lead to a failed simulation:
`--override riscvOVPsim/cpu/pk/reportExitErrors=F (Boolean) (default=F) (default) Report non-zero exit() return codes as simulator errors`

At least to my knowledge, when using Spike & riscv-pk the exit code of spike should always be the exit code of the program (wihich of course, has disadvantages, as you just mentioned.)

If this flag is enabled, it will also print the exit code to stdout, which at least allows extracting the actual value by parsing the output. I prefer this approach over handing the printing of the Exit code in the target SW (using custom syscalls) which is not even feasible when using semihosting.