tum-ei-eda/etiss #141: Instruction Pointer Updated for Illegal Instruction Codes

JoGei, 2023-11-20:
When encountering an Illegal Instruction, e.g., false opcode or target registers, the instruction pointer is still updated before the codeblock returns an exception.
When handling the exception the current Instruction Pointer is put in the MEPC such that software can identify the illegal instruction. In ETISS, this is not accurate because the instruction pointer was updated in the illegal instruction with its default increment (+2/+4) which leads to the MEPC being set not to the illegal instruction, but its default successor.

Here is an Example:
https://github.com/tum-ei-eda/etiss/blob/0d215c87a794a7979f4d728deeeeee0408b8f169/ArchImpl/RV32IMACFD/RV32IMACFD_RV32ICInstr.cpp#L541-L550

Possible fixes, I could think of:

- only update instruction pointer on legal instructions
- store the illegal instruction pointer somewhere to consume by exception handling

wysiwyng, 2023-11-20:
this should be fixed in #127, please test.

JoGei, 2023-11-20:
Yes looks good. Will close when #127 is upstream.

closed 2024-01-18 by wysiwyng