tum-ei-eda/etiss #148: CI: Broken after removing 'examples/'  8f2fb89

JoGei, 2024-06-06:
After commit 8f2fb89 removed the `examples/` CI was not updated and fails.

jokap11, 2024-06-09:
Do you think this is a suitable solution? https://github.com/tum-ei-eda/etiss/pull/138/commits/0b8ad8ad0158062950a813b8b1485b77b1894e93 
Removing all example parts?
Or should the CI file load the submodule?

PhilippvK, 2024-06-09:
The etiss riscv tests moved to https://github.com/tum-ei-eda/etiss_riscv_examples.

(@wysiwyng)

To fix the CI in the `etiss` repo we could:
1. Drop the CI jobs all together (not recommended)
2. Build new CI jobs in this repository based on `etiss_riscv_examples`.
3. Add CI to `etiss_riscv_examples` repository and trigger it from the `etiss` repo (if feasible)

JoGei, 2024-06-10:
> Do you think this is a suitable solution? [0b8ad8a](https://github.com/tum-ei-eda/etiss/commit/0b8ad8ad0158062950a813b8b1485b77b1894e93) Removing all example parts? Or should the CI file load the submodule?

No, not really. I think the examples and the CI tests were quite helpful as a form of "sanity tests". So I would try to keep them.

wysiwyng, 2024-06-11:
I also think 2.b) is the best option, I am rewriting the ci file. We could also add https://github.com/tum-ei-eda/etiss_riscv_tests here to have actual functional tests. These should then only report the status, not fail the CI when tests don't pass.

PR #149 (2024-07-04): Use outsourced examples in CI
Fixes #148

LLVM 11.1 was also allowed for ETISS to allow LLVM installation via `apt` on Ubuntu.

The old performance comparisons are omitted for now, they could be added back if desired. Dhrystone benchmarks are still run, the results of which are available in the GitHub Actions artifacts.
closed 2024-07-04 by wysiwyng