tum-ei-eda/etiss #173: LLVMJIT: Support newer LLVM versions

PhilippvK, 2025-09-24:
We have a dependency on LLVM 11.0.0 (2020) for the LLVMJIT.

Especially since there are no builds available for LLVM 11 on Ubuntu 22.04+ in the package managers, it would be great to move to a newer LLVM version...

I remember some conversation with Karsten who mentioned that newer LLVM Releases will not work easily with ETISS, but I guess I can just try out and see what needs to be patched.

JoGei, 2025-09-30:
LLVM compile times are rather slow, it seems. I don't think upgrading LLVm versions alone will fix this.
I found this work, TPDE, for a fast LLVM-JIT engine from TUM CS colleagues: 

- https://github.com/tpde2/tpde
- Pre-print article: https://arxiv.org/abs/2505.22610

Would it make sense to have a look at that, in case we refactor the LLVMJIT in ETISS?

PhilippvK, 2025-09-30:
@JoGei my priority would be to drop the need for LLVM 11 which is not available for Ubuntu 24.04+.

Looking into performance improvements has minor relevance, at LLVMJIT is rarely used and TCCJIT outperforms GCCJIT and LLVMJIT for almost any workload < 10M instructions. The main advantage of LLVMJIT vs. GGCJIT is probably that it does not need to spawn shell subprocesses (aka. less overhead)